sparkfun/SparkFun_MLX90640_Arduino_Example #1: Not working with Arduino Due

ti-89, 2018-06-14:
Hi. Tested your code with **Arduino Due**. I2C problem while reading EEPROM dump and other large registers. In **MLX90640_I2CRead()**, its seems like the **startAddress** (which will be updated every **I2C_BUFFER_LENGTH**) will not be recognized by the MLX90640. If you print out the resulting EEPROM dump (**eeMLX90640**), the first 32-bytes will be repeated over and over. Any suggestions?

closed 2018-06-15 by ti-89

nseidle, 2018-06-25:
A couple things come to mind:

* Are you using a bus pirate? The nomenclature looks that way. We've had extremely bad luck using the pirate to trouble shoot I2C calls. Use a logic analyzer instead. Namely, the way BP does multiple byte reads is super odd and we haven't figured out how to get it to read reliably.

* The 'bug' you describe seems odd. I haven't used the Due with the MLX90640 but what you describe is what I saw when I was first designing the library. My problem was that the MLX90640 does not auto-increment the address pointer. See [this issue](https://github.com/melexis/mlx90640-library/issues/2). Be sure you're writing the address to read from before doing a read.

ti-89, 2018-07-15:
Hi There
- It’s not a bus pirate but a similar product. 
- The issue is definitively related to the Wire implementation in the Due. The exact same code runs fine with a Teensy 3.2 but fails with "Parameter extraction failed" or similar errors on the Due.

cimba007, 2018-09-28:
I got the same problem on an https://github.com/rogerclarkmelbourne/Arduino_STM32 .. my EEPROM reads like

```
8 
A7 799F 0 2061 4 320 3E0 81F 2C48 1186 499 0 188 0 0 8 A7 799F 0 2061 4 320 3E0 81F 2C48 1186 499 0 188 0 0 8 
A7 799F 0 2061 4 320 3E0 81F 2C48 1186 499 0 188 0 0 8 A7 799F 0 2061 4 320 3E0 81F 2C48 1186 499 0 188 0 0 8 
A7 799F 0 2061 4 320 3E0 81F 2C48 1186 499 0 188 0 0 8 A7 799F 0 2061 4 320 3E0 81F 2C48 1186 499 0 188 0 0 8 
A7 799F 0 2061 4 320 3E0 81F 2C48 1186 499 0 188 0 0 8 A7 799F 0 2061 4 320 3E0 81F 2C48 1186 499 0 188 0 0 8 
A7 799F 0 2061 4 320 3E0 81F 2C48 1186 499 0 188 0 0 8 A7 799F 0 2061 4 320 3E0 81F 2C48 1186 499 0 188 0 0 8 
A7 799F 0 2061 4 320 3E0 81F 2C48 1186 499 0 188 0 0 8 A7 799F 0 2061 4 320 3E0 81F 2C48 1186 499 0 188 0 0 8 
A7 799F 0 2061 4 320 3E0 81F 2C48 1186 499 0 188 0 0 8 A7 799F 0 2061 4 320 3E0 81F 2C48 1186 499 0 188 0 0 8 
A7 799F 0 2061 4 320 3E0 81F 2C48 1186 499 0 188 0 0 8 A7 799F 0 2061 4 320 3E0 81F 2C48 1186 499 0 188 0 0 8 
A7 799F 0 2061 4 320 3E0 81F 2C48 1186 499 0 188 0 0 8 A7 799F 0 2061 4 320 3E0 81F 2C48 1186 499 0 188 0 0 8 
A7 799F 0 2061 4 320 3E0 81F 2C48 1186 499 0 188 0 0 8 A7 799F 0 2061 4 320 3E0 81F 2C48 1186 499 0 188 0 0 8 
A7 799F 0 2061 4 320 3E0 81F 2C48 1186 499 0 188 0 0 8 A7 799F 0 2061 4 320 3E0 81F 2C48 1186 499 0 188 0 0 8 
A7 799F 0 2061 4 320 3E0 81F 2C48 1186 499 0 188 0 0 8 A7 799F 0 2061 4 320 3E0 81F 2C48 1186 499 0 188 0 0 8 
A7 799F 0 2061 4 320 3E0 81F 2C48 1186 499 0 188 0 0 8 A7 799F 0 2061 4 320 3E0 81F 2C48 1186 499 0 188 0 0 8 
A7 799F 0 2061 4 320 3E0 81F 2C48 1186 499 0 188 0 0 8 A7 799F 0 2061 4 320 3E0 81F 2C48 1186 499 0 188 0 0 8 
A7 799F 0 2061 4 320 3E0 81F 2C48 1186 499 0 188 0 0 8 A7 799F 0 2061 4 320 3E0 81F 2C48 1186 499 0 188 0 0 8 
A7 799F 0 2061 4 320 3E0 81F 2C48 1186 499 0 188 0 0 8 A7 799F 0 2061 4 320 3E0 81F 2C48 1186 499 0 188 0 0 8 
A7 799F 0 2061 4 320 3E0 81F 2C48 1186 499 0 188 0 0 8 A7 799F 0 2061 4 320 3E0 81F 2C48 1186 499 0 188 0 0 8 
A7 799F 0 2061 4 320 3E0 81F 2C48 1186 499 0 188 0 0 8 A7 799F 0 2061 4 320 3E0 81F 2C48 1186 499 0 188 0 0 8 
A7 799F 0 2061 4 320 3E0 81F 2C48 1186 499 0 188 0 0 8 A7 799F 0 2061 4 320 3E0 81F 2C48 1186 499 0 188 0 0 8 
A7 799F 0 2061 4 320 3E0 81F 2C48 1186 499 0 188 0 0 8 A7 799F 0 2061 4 320 3E0 81F 2C48 1186 499 0 188 0 0 8 
A7 799F 0 2061 4 320 3E0 81F 2C48 1186 499 0 188 0 0 8 A7 799F 0 2061 4 320 3E0 81F 2C48 1186 499 0 188 0 0 8 
A7 799F 0 2061 4 320 3E0 81F 2C48 1186 499 0 188 0 0 8 A7 799F 0 2061 4 320 3E0 81F 2C48 1186 499 0 188 0 0 8 
A7 799F 0 2061 4 320 3E0 81F 2C48 1186 499 0 188 0 0 8 A7 799F 0 2061 4 320 3E0 81F 2C48 1186 499 0 188 0 0 8 
A7 799F 0 2061 4 320 3E0 81F 2C48 1186 499 0 188 0 0 8 A7 799F 0 2061 4 320 3E0 81F 2C48 1186 499 0 188 0 0 8 
A7 799F 0 2061 4 320 3E0 81F 2C48 1186 499 0 188 0 0 8 A7 799F 0 2061 4 320 3E0 81F 2C48 1186 499 0 188 0 0 8 
A7 799F 0 2061 4 320 3E0 81F 2C48 1186 499 0 188 0 0 8 A7 799F 0 2061 4 320 3E0 81F 2C48 1186 499 0 188 0 
```

This is to be expected as the startAddress does not get autoincremented

[image]

(The address does not change!)