sparkfun/SparkFun_MLX90640_Arduino_Example #3: MLX90640 not detected after couple of minutes

sebmorales, 2018-06-29:
Hi, 
Not sure what is going on, or if this is a problem with the library. I tied the code and connected to the processing sketch and it worked great for a couple of minutes. Then it gave me an error saying no ack read. After reseting the error change to:"MLX90640 not detected at default I2C addres. Please check wiring. Freezing."

The sensor started working again after a couple minutes but the same thing happened then. This repeated for a handful of times. Works for a couple of minutes then I2C freezes. 

Now the sensor won't start at all.

I am using Qwiic IR Array MLX90640 with a teensy 3.5. Nothing else just sample code.  

Any ideas of what I might try?
Thanks

sebmorales, 2018-06-30:
I tied it again yesterday and it was working with no problems... Not sure if something was causing too much noise for the i2c communication...

nseidle, 2018-07-02:
Hmm, finicky hardware issue. Double check the solder connections on our board. If they look solid (they should) then I have to begin to point at the MLX90640 sensor itself. I have not experience this before. Please check back in and let us know what you find!

sebmorales, 2018-07-04:
The solders looked good and I reflowed them just to make sure. I still have the same problem. Sometimes it works well, then will stop for a while then start working again, stopping suddenly.

"MLX90640 not detected at default I2C addres. Please check wiring. Freezing."

nseidle, 2018-07-06:
I assume you're powering it at 3.3V?

sebmorales, 2018-07-06:
Yes, I am connecting to the following pins: 
[image]

tsb4, 2020-07-20:
I had the same problem here, with an ESP32. Did you get solved the problem?