sparkfun/SparkFun_MLX90640_Arduino_Example #4: Parameter extraction failed

ysshah95, 2018-07-05:
Hello,
I have been using MLX90640 with Arduino Due and after uploading the Basic_Reading program I get a message in serial monitor saying "Parameter extraction failed" and the pixel values are displayed as nanC. 

Any ideas what I might try?

Thank You
Yash Shah

nseidle, 2018-07-06:
Hmm. That is odd. I'm not entirely sure but let's eliminate a handful of things:

* What MLX90640 board are you using?
* How are you powering it?
* Do you have resistors on the I2C lines?

It sounds kind of like I2C is not working but I would have expected the code to fail [earlier](https://github.com/sparkfun/SparkFun_MLX90640_Arduino_Example/blob/master/Firmware/Example1_BasicReadings/Example1_BasicReadings.ino#L50).

ysshah95, 2018-07-06:
- I am using Qwiic IR Array - MLX90640 110° FOV by SparkFun [https://www.sparkfun.com/products/14768]
- I am Powering Arduino Due with micro USB.
- Yeah, I have 2 pull up resistors (2.2k each) on I2C  lines.

nseidle, 2018-07-06:
Ok - The Qwiic board already has pull up resistors so you shouldn't need additional ones (shouldn't hurt though).  Have you got a logic analyzer? Can you confirm the Due's I2C port is performing as expected?

ysshah95, 2018-07-09:
I don't have the logic analyzer but I will somehow confirm the working of Due's I2C port. Do you think changing the controller to Teensy 3.5 would be the perfect option if I2C of Due is not working?

ysshah95, 2018-07-09:
I took a help from this Arduino Forum (https://playground.arduino.cc/Main/I2cScanner) to check the working of I2C of Due, and to my surprise, the I2C of Due is working and it displays "I2C device found at address 0x33  !", message every 5 seconds.

nseidle, 2018-07-09:
Good - that means the sensor is at least acknowledging I2C traffic.

Changing the controller to a Teensy would eliminate one unknown. I'll see if I can get a Due and test things out.

ysshah95, 2018-07-09:
Okay, that would be a great help. Thank you so much.

hello-error, 2018-07-12:
Same problem as above with an ESP32

ysshah95, 2018-07-12:
Okay. So, is it recommended to change the sensor?

hello-error, 2018-07-12:
No, the sensor is perfectly fine, so is the code here. I haven't tweaked it yet. Will get back with results for ESP32 Arduino .. Meanwhile this might help you with a raspberry pi [https://github.com/pimoroni/mlx90640-library](https://github.com/pimoroni/mlx90640-library)

ti-89, 2018-07-15:
The issue is definitively related to the Wire implementation in the Due. I had the same problem but switched to a Teensy 3.2. I didn't investigate the problem to the end but you can search the web for Due & I2C issues.

nseidle, 2018-08-06:
Whew. I've spent some time on this and cannot reach a resolution. The Due I2C library and hardware seem buggy.

[image]

Above is an image of the Due glitching the clock line. Wire and Wire1 ports both exhibit this bad behavior. 

Attached is the code I was using to test. This is as far as I can take it at the moment.

nseidle, 2018-08-07:
Hmm. Here's the code again.

[Example3_DueExample.zip](https://github.com/sparkfun/SparkFun_MLX90640_Arduino_Example/files/2268321/Example3_DueExample.zip)

ysshah95, 2018-08-10:
Thank you so much for the help. I really appreciate your effort. I recently changed my controller from Due to Teensy 3.5, and the library works fine. 

But, I also tried the latest Due example you attached in the last comment, but still, it doesn't work. I think it gets stuck in one of the function "MLX90640_DumpEE". I have attached the screenshot of the terminal.

[image]

nseidle, 2018-08-10:
Indeed, the Due Example I attached does not work, it's only an example of the changes that were started in an attempt to change the lib. The example can be used to inspect the clock glitches from the Due.

closed 2018-09-26 by ysshah95