sparkfun/SparkFun_MLX90640_Arduino_Example #15: Working with ESP8266

vishalkothari, 2019-05-14:
Hi,

I am trying to use MLX90640 with ESP8266 and run Example1_BasicReadings.
But, I have not been able to get I2C communication to work. 

Call to MLX90640_DumpEE > MLX90640_I2CRead > Wire.endTransmission(false) returns 2 or 3. Sometimes call to MLX90640_DumpEE hangs.

I am using frequency 100KHz. I have seen issue #2 and https://github.com/melexis/mlx90640-library/issues/13. 

Below is the output I get.

```
MLX90640 online!
No ack read
Pixel 0: 884.73C
Pixel 1: 0.00C
Pixel 2: 573.09C
Pixel 3: 0.00C
Pixel 4: nanC
Pixel 5: 0.00C
Pixel 6: nanC
Pixel 7: 0.00C
Pixel 8: nanC
Pixel 9: 0.00C
No ack read
No ack read
```

I have also modified core_esp8266_si2c.cpp > twi_readFrom function to handle len=0
```
unsigned char twi_readFrom(unsigned char address, unsigned char* buf,
 unsigned int len, unsigned char sendStop){
   ...
  for(i=0; i<(len-1); i++) buf[i] = twi_read_byte(false);
  buf[len-1] = twi_read_byte(true); //original
  ....
}
```
to 
```
unsigned char twi_readFrom(unsigned char address, unsigned char* buf,
 unsigned int len, unsigned char sendStop){
   ...
  if (len > 0) { 
    for(i=0; i<(len-1); i++) buf[i] = twi_read_byte(false);
    buf[len-1] = twi_read_byte(true); 
  }
  ....
}
```

Please let me know if you have any pointers to solve this. 
@nburnett, I saw your comment that you were able to connect MLX90640 with ESP8266, can you please share more information about the same.

Thanks in advance.

Regards,
Vishal Kothari

tmsd2001, 2019-05-29:
maybe it will help you, with the ESP32 I can use all the sketches with almost no changes.
I put only #include<Arduino.h> into the MLX90640_I2C_Driver.ccp
I can use 100 KHz or 400 KHz
That's my I2C setting:
Wire.begin(5, 4); // I2C connect to D5, D4
Wire.setClock(400000);

vishalkothari, 2019-06-04:
@tmsd2001 , Thank you. 
I switched to ESP32 and it seems to be ok.
I am facing another problem that Pixel 22 always has value NaN. Tracing this further, it seems irData calculated in MLX90640_API.cpp has a large negative value like -6874.43. irData for other pixels is in -10 to -20 range. I will post another issue for this.

closed 2019-06-04 by vishalkothari

oficjalniebktv, 2024-02-03:
cześć, masz już może poprawiony kod  MLX90640 z ESP8266 i czy odczyt  jest formie obrazu na stronie web czy tylko w formie liczb?