sparkfun/SparkFun_MLX90640_Arduino_Example #18: 'Serial' was not declared in this scope

h-Naima-s, 2019-09-28:
Getting the following error message when uploading this code on arduino:

sketch\MLX90640_I2C_Driver.cpp: In function 'int MLX90640_I2CRead(uint8_t, unsigned int, unsigned int, uint16_t*)':

MLX90640_I2C_Driver.cpp:48:7: error: 'Serial' was not declared in this scope

       Serial.println("No ack read");

       ^

sketch\MLX90640_I2C_Driver.cpp: In function 'int MLX90640_I2CWrite(uint8_t, unsigned int, uint16_t)':

MLX90640_I2C_Driver.cpp:95:5: error: 'Serial' was not declared in this scope

     Serial.println("Error: Sensor did not ack");

     ^

exit status 1
'Serial' was not declared in this scope

qTp, 2019-11-13:
Has a solution been found for this problem?
I get the same error message, with an Adruino Leonardo board.

h-Naima-s, 2019-11-13:
Yes! so you need to run the code as it is on Processing app. It will show you 2 ports with error message, and out of those 2 will show you one as available. you need to then go to myPort on line 59 and change the port number to one that is showing available. 
This was what going wrong with me.

qTp, 2019-11-13:
thanks for the fast answer. 
i've found another soultion:
       In the MLX90640_I2C_Driver.h, i included the "Arduino.h" header file.

santaimpersonator, 2019-12-28:
Seems resolved. Closing issue.

closed 2019-12-28 by santaimpersonator