sparkfun/SparkFun_MLX90640_Arduino_Example #19: "Parameter extraction failed" Error

RawanAlharbi, 2019-11-09:
Hello,

I tried to upload the example to a Teensy 3.6 and it successfully uploads with no errors. However, in the serial monitor, I get "Parameter extraction failed":

> MLX90640 IR Array Example
> MLX90640 online!
> Parameter extraction failed
> Pixel 0: nanC
> Pixel 1: nanC
> Pixel 2: nanC
> Pixel 3: nanC
> Pixel 4: nanC
> Pixel 5: nanC
> Pixel 6: nanC
> Pixel 7: nanC
> Pixel 8: nanC
> Pixel 9: nanC
> Pixel 0: nanC
> Pixel 1: nanC

 Any ideas on the source of the issue?

RawanAlharbi, 2019-11-09:
I fixed it. You have to update the MLX90640 library. The latest can be found [here](https://github.com/melexis/mlx90640-library). 

basically just replace:
- MLX90640_API.cpp
- MLX90640_API.h

mcnpower, 2019-11-21:
Hi Rawan, 
I received the same error with teensy 3.5. When I changed the .cpp and .h files from your link, code not building for teensy 3.5.  Do I need teensy 3.6? Is it important 3.5 or 3.6? Could you share full project files? Thanks.

nunomiguelferreira, 2020-01-25:
I have the same issue when using a SAMD21 based system. But I always get a status of -4 with parameter extration, although I get values on the temperature. The same camera works fine with an ESP8266

Allen5435, 2020-03-17:
I use ESP32 changed API and solve the problem.
It looks like melexis change the storage offset of parameter and caused error.

mcnpower, 2020-03-17:
I found the problem. Sparkfun breakout was problematic. The pins of the sensor are short-circuited. I heated up the solders and slid it up. Now the sensor is working. They wrote this on their forums. Some cards were produced incorrectly. I bought it from Digikey. Notice, this process is difficult and requires professionalism.

tomeroto, 2021-01-24:
> I found the problem. Sparkfun breakout was problematic. The pins of the sensor are short-circuited. I heated up the solders and slid it up. Now the sensor is working. They wrote this on their [...]

can you give me youtube video how you fix it?

sachinbs1995, 2021-08-10:
facing the same issue in this API also the same error.

calvfoo, 2021-08-29:
ok, got it working with codes from here
https://github.com/Samox1/ESP_Thermal_Camera_WebServer

use Arduino IDE standard ESP32 board

MLX90640 from here
https://www.aliexpress.com/item/32962087015.html?spm=a2g0s.9042311.0.0.65b74c4dronS9H