sparkfun/SparkFun_MLX90640_Arduino_Example #25: Result is very dirty, any idea?

Dev-Damon, 2022-02-16:
[image]
[image]
[image]

I'm using ESP8266. but I'm working on the official FreeRTOS SDK of esp8266.
I'm just wondering what's happening in my sensor,...

I'm sure there is no problem with i2c sensor.
I've checked status register and ctrl register every reading sensor. it should fine.
ctrl reg : 0x1901
- 2Hz
I've tried with 0.5Hz to 4Hz but it works like same.

Have you guys ever been to this situation? 

Thanks in advanced.

LBuydens, 2022-02-16:
It looks like an image distortion one can get if the power decoupling capacitor is not present or not close enough to the sensor.

Dev-Damon, 2022-02-17:
> It looks like an image distortion one can get if the power decoupling capacitor is not present or not close enough to the sensor.

Thanks for answering.
I've added 2 capacitor which are 100nF & 10uF like datasheets says, but it works same :(

I should try with another MCU... not ESP8266

I have total 4 sensor, 2 are FOV55, other Two are FOV 110, all of them are MLX90640, but works same.
I have a huge doubt in hardware, not the code and the sensors. but I don't really don't know what the problem is.

Dev-Damon, 2023-10-13:
> > > It looks like an image distortion one can get if the power decoupling capacitor is not present or not close enough to the sensor.
> > > ------ edited
> > > Thanks for answering.
> > 
> > 
> > I added capacitors so the pattern disappears. but still chess pattern remains. I have a huge doubt in hardware, not the code and the sensors. but I don't really don't know what the problem is.
> > [image] [image]
> 
> Did you solved it?

No. I just figured out with after process.... that is the maximum performance of the sensor.
I checked sensor with official software of melexis,(called official manager of melexis) but it couldnt solved. just used median filter, and IIR filter.