sparkfun/SparkFun_MLX90640_Arduino_Example #26: Sparkfun MLX90640 i2c address issue

frank984, 2022-10-25:
I connected two Sparkfun IR arrays to a Raspberry Pi 3a+ using a PCA9548A multiplexer:
[image]
Sometimes the code used read the IP of the sensors correctly:
[image]
Sometimes not, with the following string:
[image]

The same thing happens when I use only a module connected directly to the Raspberry, without the multiplexer: without any variations hw or sw, sometimes the i2C address of the module is read correctly, sometimes not
How can I solve the problem?

TurnerRussell50, 2022-10-26:
it's the i2c pullup jumpers; cut one of the sensors' traces on the board rear like so https://learn.sparkfun.com/tutorials/qwiic-ir-array-mlx90640-hookup-guide?_ga=2.150089077.238313185.1666643760-1841065663.1665480869&_gac=1.149695044.1665786185.CjwKCAjwkaSaBhA4EiwALBgQaA0DWIJgJYLZ5bEYuxxY6utz93O-6RlGy3eIyQJ-uKGoKDj382tI1xoCH9UQAvD_BwE#hardware-overview (leave one set intact)

frank984, 2022-11-04:
Even if I cut off all the extra sensors' I2c pullup jumpers, the modules do not have a '0x33' address.
I solved the problem by desoldering the mlx90640 from the board:
[image]
Now the three sensors have the same '0x33' address:
[image]